robotika/osgar #11: Decide on a workflow

zwn, 2017-06-21:
We should have a file in the repository describing how development is done in this repository. Something similar to [git flow](http://nvie.com/posts/a-successful-git-branching-model/), [github flow](http://scottchacon.com/2011/08/31/github-flow.html), [gitlab flow](https://docs.gitlab.com/ce/workflow/gitlab_flow.html) or [bitbucket flow](https://www.atlassian.com/blog/archives/simple-git-workflow-simple) (or any other flows that exist).

We can flesh out the details in this issue and the summarize it to a file.

zwn, 2017-06-21:
I'll start.

* master is never rebased or otherwise edited and always works
  - compiles
  - all tests pass
* all development is done on branches
* branches and PR are as contained and small as possible and reasonable
* each PR is reviewed by a member of @robotika
* PR are reviewed as one change
* before merging to master branches are rebased and squashed

m3d, 2017-06-26:
OK, I am not 100% sure about _all branches are based on master and all PRs are to master_, but it makes clear preference. And we want to make the cycle short.

zwn, 2017-06-27:
Let's say those are the defaults (or preference) and not the law. Anything can be done differently if there are strong arguments for doing so.

m3d, 2018-03-15:
Should not we put your points in `README.md` or somewhere else?

zwn, 2018-03-15:
Maybe CONTRIBUTING.md would be a better choice?

https://blog.github.com/2012-09-17-contributing-guidelines/

zwn, 2018-11-07:
We have contributing.md in the root of the project.

closed 2018-11-07 by zwn