robotika/osgar #23: Change timestamp implementation to round trip safely to/from log

zwn, 2017-07-24:
Using `time.time()` serialized as a string is probably not a good idea since the round trip is not a lossless operation.

zwn, 2017-07-26:
In video files, time is usually handled by defining a "time base" (how many units go in one second) and then using this time base to define time. That defines time as rational number allowing trade-off between resolution and size. Time base could be even set in config.

zwn, 2017-07-28:
I am missing my previous comment on this issue, @m3d do you have it maybe in an email notification?

m3d, 2017-07-30:
There was a comment in private e-mail, but ... ignore. Yes, I do not want to use `time.time()` for long term and yes, I prefer integer indexes (with conversion multiplier). Ideal is the fastest source as I do not see much point in 64bit time stamps used in ADTF for example.

zwn, 2017-07-31:
Integer indexes with a multiplier is what is used in multimedia containers that we are considering for log storage. However, ffmpeg also use 64bit numbers in their API to get current time (see [av_gettime](http://dranger.com/ffmpeg/functions.html#av_gettime)). It returns microseconds. The time inside the file uses [AVRational](http://dranger.com/ffmpeg/data.html#AVRational) and the time base is get from the file.

zwn, 2017-08-01:
ffmpeg uses `int64_t` for `pts` (aka Presentation TimeStamp) in [AVFrame][1] and in [AVPacket][2] but it is still relative to `time_base`.

[1]: https://github.com/mikeboers/PyAV/blob/95f5bb72d69fe8096cebb00d5b54cb7c1aca0661/include/libavcodec/avcodec.pxd#L254
[2]: https://github.com/mikeboers/PyAV/blob/95f5bb72d69fe8096cebb00d5b54cb7c1aca0661/include/libavcodec/avcodec.pxd#L218

m3d, 2018-03-15:
Are we happy with `osgar.lib.logger` so we could close this issue?

zwn, 2018-03-15:
Yes!

closed 2018-03-15 by zwn