robotika/osgar #84: Logging of large data blocks

m3d, 2018-09-21:
Starting with camera #82 there is a need for larger data blocks than 0xFFFF bytes. I would propose to implement serialization this way:
* all data smaller than 0xFFFF would be as for now
* large blocks would be stored in 0xFFFF pieces, with identical timestamp and channel
* block of size multiple 0xFFFF would be followed by a block of zero length.

If you would agree I would implement it ASAP (due close to SICK Robot Day dead-line).
thanks m.

m3d, 2018-09-22:
An alternative would be to use 3-bytes length and increase block limit to 16MB
```
>>> 0x1000000
16777216
```
and use only 255 options for `streamID`.

Or we could use only one bit as `overflow flag` from `streamID`, keep 15bit for `streamID` and merge incomplete messages on demand?

zwn, 2018-09-24:
I would go for simplicity/readability - just use as many bytes as needed, fixed size. The "big boys" are using variable length encodings for this but I don't think we need it. Are the space savings worth the complicated implementation? I think they are not.

m3d, 2018-09-24:
Hmm, I started to implement the first option - somehow it feels bad to waste so much for small (typical) messages :(. Thanks for your comment anyway. We may drop it - it should be ready in less than 1hour. I also wanted to keep backward compatibility.

zwn, 2018-09-24:
Feel free to go ahead. We'll see how complicated the code is.

In general I try not to have encoding inside encoding inside... you get the point. Splitting the data payload over several packets is such overlayed encoding. With several additions like that (taking this as a precedent) it can get complicated very quickly.

m3d, 2018-09-24:
OK, I understand and partially agree, but ... it is like with timestamps, where I do not want to use 64bit NTP, which is repeated many many times. You will see. It is not perfect ...

m3d, 2018-09-26:
p.s. it turned out that the unexpectedly large image block was probably error (attempt to stream video?) and individual images fit into 65534 limit (print of data sizes only):
```
m:\git\osgar>python osgar\logger.py d:\md\osgar\logs\jk\eduroLogs_180925b\eduro-180925_211606.log --times --stream camera.raw
0:00:00.091831 11 62943
0:00:00.223641 11 62943
0:00:00.354597 11 62943
0:00:00.487010 11 62943
0:00:00.760808 11 62943
0:00:00.891707 11 64393
0:00:01.022984 11 62943
...
``` 
Nevertheless it should be able to log a large block if that "accidentally" happen (see #85 ).

zwn, 2018-11-07:
Is this ready to be closed?

m3d, 2018-11-07:
Yes.

closed 2018-11-07 by m3d