robotika/osgar #86: Reporting/logging module errors

m3d, 2018-09-26:
The modules can fail to do their job, and it would be nice to see it afterwards. If the error is reported only on stderr you may overlook it. We can use our "system stream 0" to dump the problem there (actually I already once had to do it when "hacking" I2C failure on boat (https://github.com/robotika/osgar/commit/273d9be60959fb113230660e0997fa2463c6bcbd) but I would prefer some nicer unified way ... any suggestion?

`bus.report_error(e)`?
```
                   except OSError as e:
                        print(e)
                        with self.bus.logger.lock:
                            self.bus.logger.write(0, bytes(str(e), encoding='ascii'))
```
would be now
```
                   except OSError as e:
                        self.bus.report_error(e)
```
?

m3d, 2018-10-01:
preliminary working version is at https://github.com/robotika/osgar/commit/b94c1a35b8e70735f2b061b1465cbe57fd01bdc8

zwn, 2018-11-07:
are you happy with it?

m3d, 2018-11-07:
I guess yes - I have not seen it for a while. Let's close it.

closed 2018-11-07 by m3d