robotika/osgar #110: Should bus.publish() raise BusShutdownException?

m3d, 2018-11-17:
I was suprised that following code was never ending:
```
    def run(self):
        try:
            while True:
                self.bus.publish('tick', None)
                self.bus.sleep(self.sleep_time)
        except BusShutdownException:
            pass
```
Then I realized that there should be `self.bus.is_alive()` instead but ... do you remember why? Now I would propose to have both ... as far as I remember the `is_alive()` is for "boundary" nodes which have only external inputs i.e. they do not use `listen()` and the external inputs could be very rare ... ???

zwn, 2018-11-19:
I don't remember. Raising `BusShutdownException` even from `publish` seems like a good idea.

zwn, 2018-11-20:
Hmm, hmm. Interesting. So when a shutdown is requested, node is allowed to call `publish` but if all nodes were requested to shutdown, the published message is not going to be read. But it is going to be in the log file. While the `None` used to denote shutdown is not going to be in the log file. It is somewhat interesting we are still able to replay all the way to the end. 

I propose to refactor/redesign/fix this during the refactoring we have been discussing lately where nodes would be required to inherit from a common base class.