robotika/osgar #245: Call of self.buf.shutdown() kills only caller

m3d, 2019-10-06:
i.e. it does not distribute "shutdown request" to other threads/modules ...

zwn, 2019-10-07:
That's right. I was wondering why we could go out so long without noticing and we are only shutting down the application which takes down everything so it has not been an issue so far. How did you notice? What was the use case?

To fix it, the node wanting to initiate shutdown would have to have a means of contacting the other nodes. Until recently there was no such thing but now I think 
https://github.com/robotika/osgar/blob/44c581214bf7e455cba62cc3e1fa322a9b1edf7a/osgar/record.py#L17-L19
could be useful for this. I think it will somehow fit into my work on logging of shutdown. Currently the only shared stuff among `BusHandler` instances is [logger](https://github.com/robotika/osgar/blob/44c581214bf7e455cba62cc3e1fa322a9b1edf7a/osgar/logger.py#L48). Maybe the notion of shutting down should be part of it? Need to think about it more (mumbling to myself).

m3d, 2019-10-07:
I noticed some indication before (LoRa should be able to kill the program, but did not in SubT System Track), and recently I modified recorder for "infinite recording" just to wait for all modules to finish and
`ls -lh | python3 -m osgar.record subt/config/stdout2ros.json` never finished, so I had to add artificial timeout :(. The cascade shutdown is non-trivial ... and is partially discussed elsewhere(?)

zwn, 2019-10-07:
>  The cascade shutdown is non-trivial ... and is partially discussed elsewhere(?)

I think it is enough what the SIGINT handler does:
https://github.com/robotika/osgar/blob/44c581214bf7e455cba62cc3e1fa322a9b1edf7a/osgar/record.py#L60-L65

It is just that currently the nodes don't have a pointer to the instance of Recorder.

zwn, 2019-10-07:
BTW: I couldn't find stdout2ros.json anywhere. Where is it and what it does it do?

zwn, 2019-10-07:
Let me take back that `request_stop` call. Now that I think about it, it is even simpler - it should be enough to just call `recorder.stop_requested.set()` which is even thread-safe. The main thread is blocked on this (with or without timeout) so once it is set, it will take down the whole system correctly.

m3d, 2019-10-07:
https://github.com/robotika/osgar/blob/feature/virtual-ignition/subt/config/stdout2ros.json

m3d, 2019-10-07:
> BTW: I couldn't find stdout2ros.json anywhere. Where is it and what it does it do?

https://github.com/robotika/osgar/blob/feature/virtual-ignition/subt/config/stdout2ros.json