robotika/osgar #246: Log recovery

m3d, 2019-10-07:
It would be nice to be able to recover OSGAR log file from missing header ... say that the user has only 10MB of the end of the logfile. The log is dense and there are no "transaction points". Actually even hint how to do it manually would be nice. Note, that I have another logfile with header, but the order of channels does not have to be identical.

zwn, 2019-10-07:
I've done such stuff for work. The first thing I would try would be just start parsing at the first byte and see if you get meaningful data, if not, start at byte+1. You can start for example with timestamps or sizes of the messages - sizes are especially nice because they allow you to cheaply validate the rest of the file - just parse a size and try to find next packet using this size etc. That should be cheap enough to do it byte by byte for the whole file.

m3d, 2019-10-07:
Thanks - you are right that because if have defined internal structure of messages (`msgpack`) we should be able to do what you suggest :).
In meantime I used some repeating string (here "cmd_vel" and computed offset from it -- to strip begin and end and mix it with artificially generated header:

```
python -m osgar.logger --stat logs\aws\ver22b\A\log\77001f58-e8ba-11e9-a92b-7aa097a03f05\osgar.log.new
 0    sys       3084 |    24 |   0.0Hz
 1  name1      17891 |  2690 |   1.9Hz
 2  name2     158763 | 15003 |  10.9Hz
 3  name3        202 |     1 |   0.0Hz
 4  name4     510136 | 15004 |  10.9Hz
 5  name5       2261 |    13 |   0.0Hz
 6  name6          0 |     0 |   0.0Hz
 7  name7        104 |     5 |   0.0Hz
 8  name8       3160 |  3160 |   2.3Hz
 9  name9     897352 |     5 |   0.0Hz
10 name10      23199 |   377 |   0.3Hz
11 name11        339 |     9 |   0.0Hz
12 name12          0 |     0 |   0.0Hz
13 name13      10582 | 10582 |   7.7Hz
14 name14      42007 | 42007 |  30.4Hz
15 name15     118624 | 15002 |  10.9Hz
16 name16     108241 | 15002 |  10.9Hz
17 name17    6489000 |  3000 |   2.2Hz
18 name18       6001 |  6001 |   4.3Hz
19 name19     152161 | 15003 |  10.9Hz
20 name20        903 |   301 |   0.2Hz
21 name21     318382 | 10972 |   8.0Hz
22 name22          0 |     0 |   0.0Hz
23 name23          0 |     0 |   0.0Hz
24 name24          0 |     0 |   0.0Hz

Total time 0:22:59.705570
```
so it was worth 23 minutes of simulation ... not bad.
Thanks (I will review this with PR after Virtual Tunnel is over)