robotika/osgar #721: SubT estimate artifact position without knowledge of depth

m3d, 2020-10-27:
There are some cases when depth data are not available (Freyja wide lenses vs. depth camera - see bug #719) or when the object is further away then 10 meters (limit for depth camera) - backpack in P1 world. It could be useful to have "preliminary estimate" in case there will be no "precise" within 2 minutes for example? Also when object is seen by Freyja side cameras the depth is not available at all.

jisa, 2020-10-30:
With a good enough local odometry, we could triangulate the artifact from consecutive images. Ideally not immediately consecutive in the stream, because a wider "baseline", the distance between the spots where the images were taken, should give us better precision.

The triangulation would not work when moving straight towards the artifact from the camera's point of view. It may be beneficial to explore the world using a zig-zag trajectory :-)

zwn, 2020-11-18:
https://www.youtube.com/watch?v=YBwHZ8yOXfc

zwn, 2020-11-18:
https://github.com/wadimkehl/ssd-6d