robotika/osgar #725: Discussion: Better control approach for kloubak drive.

AravindaDP, 2020-10-29:
I have studied 4wd drive at https://github.com/robotika/osgar/blob/master/osgar/drivers/kloubak.py#L160 and tried to understand it geometrically. But what I understood is that it's using rudimentary proportional control to bring two parts of the robot aligned to IRC (instantaneous rotation center) relative to robot's joint that connect the two parts.

In my opinion natural control dynamics of such robot is of tractor-trailer configuration where control input is expressed relative to front drive. I tried to research for suitable formal control equations for this but couldn't find any.

Are there any formal equations that would calculate passive velocities (Which would be ideal for case like SubT virtual where we require to provide rear velocity as well) for rear drive given control input for front drive?

m3d, 2020-10-29:
@AravindaDP thanks for your question. Note, that the driver you are referring to is the real robot (Kloubak K2 and K3) and they unfortunately behave differently in real and virtual worlds. For SubT solution you can have a look at #711 I was told that in CloudSim you have to use 4WD while on real K2 we took the advantage of passive wheels (see https://robotika.cz/competitions/subtchallenge/tunnel-circuit/en#190818)

AravindaDP, 2020-10-30:
@m3d Thanks for your reply.

Yes I'm aware of the differences between virtual and physical Klouback robots. And thanks for pointing to correct controller used for virtual k2.

In fact as I understood, your control method for 4wd on both is to first assume that two parts of robot are in an angle such that centers of front and rear axels lie on a circle with radius given by `desired_linear_velocity/desired_angular_velocity` and then apply linear and angular velocity corrections based on the difference between actual angle of center joint and required angle of center joint to put the two parts of robot in such a circle.

This got to me thinking of what would be the real dynamics when we use front only control and whether there is formal equations for that. (So we could calculate velocities of rear wheels that would minimize forces applied at center joint)

What I envision is something like below.

[image]

Here when we apply a mix of linear and angular velocity at front wheels we can consider that it follows a path around IRC (instantaneous rotation center) This would cause center joint also to follow path around same IRC (but with a larger radius)
Now from that we could approximate instantaneous linear velocity of center joint and based on the direction of this vector and angle of rear joint we can approximate that rear axel also to follow a circular path around IRC' (Which would lie either left or right based on whether center joint velocity vector angle point left or right relative to rear joint)
This would also need adjustment for reversing to prevent jackknifing. I think calculating forward velocities and the simply reversing velocities could prevent such problems.

I'm not sure if this is a formal control method for such drive system. And my mechanical engineering and mathematics is rusty so I might have got all of these wrong.

It's up to you to experiment and see if this improves controllability of the robot. (Specially physical one since I doubt it could help virtual counterpart since these control equations need to run as fast as possible with the sensing rate of center joint as we are approximating nonlinear motion of center joint to an instantaneous linear motion here. So for virtual k2 this would require to implement this as a gazebo plugin/as a ros controller that runs on simulation instance itself-not on solution container to see any real advantage)

Also good luck with your work on SubT challenge.

tajgr, 2020-10-30:
@AravindaDP thank for your investigation. Yes, I see your approach require very fast control. We plan an upgrade of our physical robot and that could be a possibility for new experiments like that. At least, it is inspiration for us.

jisa, 2020-10-30:
@AravindaDP Thanks for outlining your thoughts. I believe that if you finish that path of thinking, you will end up with a very similar, if not identical, result, expressed in a different reference frame.

One thing that the existing solution(s) accounts for and your proposal seems to neglect is that it is better when IRC=IRC' and front and rear halves of the robot follow the same arc. Otherwise we would introduce extra wheel slips when the two halves follow incompatible trajectories, but are forced to align, because they are physically connected.

tajgr, 2020-10-30:
> IRC=IRC'

This does not always apply. When the robot is turning (the radius of the circle trajectory is changing) IRC !=IRC'.

AravindaDP, 2020-10-30:
Yes. In the beginning IRC and IRC' would not be equal but given that we apply constant velocity, they will eventually become equal.

Assuming initial angle between two parts is 0 in fact IRC and IRC' would be in fact in opposite directions when we first apply a none zero linear velocity combined with none zero angular velocity to front section. Then given that we keep constant velocity and assuming my diagram, IRC' would goes to infinity as front section continue to turn right and rear section turns left as rear joint reaches instantaneous velocity vector at center joint. Then as front section continue to turn further, IRC' would come same side as IRC and eventually IRC and IRC' would overlap.

All this happens while effort at center joint and wheel slips at any wheel is minimized given that controller is running at sufficiently high rate to make sure above linear approximation I derived at holds true for controller interval. 
It should also be very responsive to fast changes of control inputs given that inertia of the robot is sufficiently lower and physical constrains of robot are not violated (like in place rotations which also becomes possible if angle of center joint could reach in excess of 90deg and note that here we are talking about controlling robot with respect to front axle center)

jisa, 2020-10-30:
You are both right. IRC=IRC' if and only if the robot is moving along a circle in a constant motion. Incl. straight line as an extremely large circle.

Does that mean that generic slip-free control does not exist, even if we can apply continuous control (infinitesimally small control interval)? The front half of the robot forces the joint to move along one arc, the rear half along another one and except of the constant circular motion, these arcs do not overlap. The joint cannot follow two different trajectories at the same time.

AravindaDP, 2020-10-31:
I think in my approach generic slip-free control is guaranteed if we can apply continuous control. What it propose is, apply control input to front half, then calculate velocity input we need to apply to rear half so that rear half also would cause center joint to follow same arc tangent requested by front half.