robotika/osgar #1002: unexpected keyword argument 'duration'

milanpetrik, 2025-01-14:
osgar.replay gives the error:

  File "/home/petrikm/Git/osgar/osgar/replay.py", line 54, in replay
    reader = LogReader(args.logfile, only_stream_id=inputs.keys(), duration=duration)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: LogReader.__init__() got an unexpected keyword argument 'duration'

... When I changed replay.py at line 54 commenting out the parameter 'duration', the program worked.

m3d, 2025-01-14:
thanks - this is already fixed in several separate branches, but I did not
merge it into master - I will do ASAP (tonight most probably)
thank you for your report
    Martin

m3d, 2025-01-15:
it should be solved now - thanks for your report

closed 2025-01-15 by m3d